NVlabs/ProtoMotions #101: How to train SMPL robot using motion.pt from AMASS?

damonchang23, 2025-05-13:
Hi, thank you for the great work and for releasing this impressive codebase!

I'm new to this field and still learning, so I apologize if my question seems basic.
I'm currently trying to train an SMPL robot using a motion.pt file I generated from the AMASS dataset using your preprocessing tools.

Specifically, I created the motion.pt file using the following command:
`python data/scripts/package_motion_lib.py data/yaml_files/motion_fps_smpl.yaml /media/dataset1/amass/SMPL+HG data/motions/amass.pt`

I have a few questions about how to correctly use this .pt file in training:

1. Do I need to create a new experiment YAML config (e.g., my_smpl_robot.yaml) to train using this custom motion.pt, or is there an existing config I can reuse?

2. If I do need to define a new YAML config, is there a reference format or example you would recommend following?

3. Lastly, is this the correct way to launch the training with my motion file?
`python protomotions/train_agent.py +exp=my_smpl_robot +robot=smpl +simulator=isaacgym motion_file=data/motions/amass.pt +experiment_name=how_to_train_my_smpl_robot`

Additionally, I wanted to clarify whether it is necessary to train all of the "provided algorithms" listed in the README — such as maskedmimic, full_body_tracker, amp_mlp, and ase_mlp. Are any of them required to be trained in sequence, or can I focus on just one of them (e.g., full_body_tracker) depending on my goal?

Any help or guidance would be greatly appreciated. Thank you again for this amazing project!

tesslerc, 2025-05-13:
1. The way hydra config works (and omegaconf) is that it composes the yaml files + command line arguments together. So you can override/define parameters via CLI at runtime. In this case -- as per the readme you just add the flag `motion_file=/media/dataset1/amass/SMPL+HG data/motions/amass.pt`
2. Seems it's not needed, if I understood correctly.
3. Yes

Each agent serves a certain purpose, depending on what you want to achieve.
MaskedMimic depends on the full body tracker. But the other policies are independent.

Also feel free to ask any follow up questions. Happy to help out.

damonchang23, 2025-05-14:
@tesslerc 
Thank you so much for the quick and helpful response!

That definitely clarifies how Hydra and CLI overrides work, and I now understand that I can simply pass `motion_file=...` as a flag.

That said, I still have one point of confusion regarding the `+exp=...` part.
Let’s say my goal is to train an SMPL robot using my own motion data (the .pt file I created from AMASS). In that case, I’m not sure which experiment YAML I should point to.

Should I base it on something like `+exp=full_body_tracker/transformer_flat_terrain`, and override what I need? Or is it recommended to create a new custom YAML under configs/exp/ for my specific setup?

In short: if I just want to train an SMPL robot on my motion dataset, which `+exp=...` should I start with?

Thanks again for your help — I really appreciate it!

tesslerc, 2025-05-15:
The way we have been working is that the experiment file define the agent + task you want to run.
For example, `full_body_tracker/transformer_flat_terrain` defines the task of motion tracking (Mimic task) with a flat terrain. And the policy that will solve it is PPO.

The motion file and robot are independent. You define your own robot and the data for it. The same experiment should work for any.

Which experiment to use then depends on what are you trying to achieve.
If you want to learn a deep-mimic like policy. Then what you mentioned is good.
If you want to do more of a style-based training. You can try amp. And if you want joystick-like control, then you can look at steering.

damonchang23, 2025-05-16:
Thank you very much for the detailed explanation.

I have a couple of follow-up questions regarding the capabilities of ProtoMotions:

1. Multi-person Simulation
Is it currently possible to simulate multiple agents within the same environment using ProtoMotions? I am particularly interested in scenarios where several humanoid characters are either interacting or performing motions concurrently in a shared space.

2. SMPL Mesh Integration
Is there support for rendering full SMPL or SMPL-X meshes during simulation or visualization, as opposed to using simplified body models? While I understand that the framework supports motion data based on SMPL, I am curious whether the actual visual output can reflect the detailed SMPL mesh geometry for higher fidelity.

Thank you again for your excellent work and for maintaining this valuable resource.

tesslerc, 2025-05-25:
Regarding multi-person sim, it's not supported by default but there's nothing preventing such integration.

SMPL Mesh -- the structure is identical to the standard SMPL model (same kinematic structure), just represented using a higher fidelity mesh.
I recall https://github.com/zhengyiluo/phc/ had such a model, it should be relatively simple to add instead of the capsule SMPL humanoid.
From previous experiments I recall that the mesh humanoid is slower to simulate due to the more complex collision mesh.

closed 2025-05-26 by damonchang23