NVlabs/ProtoMotions #103: Single-frame retargeting with Mink

giakhuyendihoc, 2025-05-26:
Hi, thank you for your great work! I was wondering if the system can retarget a single frame input, as this would be useful for real-time scenarios.

tesslerc, 2025-05-26:
Yes. You can see in the script that we set `t` to start negative. See [here](https://github.com/NVlabs/ProtoMotions/blob/5047e87d3fe3cd9eac75ce14745ad7a125c5f7b3/data/scripts/retargeting/mink_retarget.py#L506).
This will run a few optimization steps to get the initial pose and from there it goes step-by-step.

You can use the same approach to perform several optimization steps for single-frame, and you can also initialize it from the previous pose to ensure the retargeting is a bit more structured (depends on the previous pose etc).

tesslerc, 2025-06-04:
For real time, you probably want to re-use the last frame you had. Or even better -- to use the current frame of the robot.
Maybe IK solver parameters (objective) can be tweaked to try minimize the rotational differences from the last frame to the new one. This way the requested transitions might be smoother.