NVlabs/ProtoMotions #107: How to make USDA model from sml?

xinyiz0931, 2025-06-11:
Hi, thanks for the great work! 
I'm wondering how to make the [smpl_humanoid.usda](https://github.com/NVlabs/ProtoMotions/blob/main/protomotions/data/assets/usd/smpl_humanoid.usda) files? Is it directly converted from XML file in Isaac Sim?

tesslerc, 2025-06-20:
It's generated using the MJCF importer in IsaacSim.
I think that for this one we slightly edited the structure, however this was a long time ago and the asset converter has likely improved since.

xinyiz0931, 2025-06-23:
Hi @tesslerc Thank you very much for the reply! 

I would like to import an asset similar to the SMPL humanoid. Do you have instructions for converting XML to USDA, including parameter setups such as collision models, inertia, and mass? I tried using the MJCF converter in 4.5.0, but the generated USDA files have noticeably different structures with the [smpl_humanoid.usda](https://github.com/NVlabs/ProtoMotions/blob/main/protomotions/data/assets/usd/smpl_humanoid.usda).

Also, I noticed that in the SMPL humanoid, collisions between certain bodies are filtered. Is it necessary to disable some collisions for the Motion Tracker algorithm? Additionally, I saw that the H1 and G1 robots have  `enable_self_collisions=False` in [protomotions/simulator/isaaclab/utils/robots.py](https://github.com/NVlabs/ProtoMotions/blob/main/protomotions/simulator/isaaclab/utils/robots.py) . Is this setup required for training?

tesslerc, 2025-07-02:
Honestly it's been a while since I've converted the assets on our end, so I don't have a good guide. I'd recommend following the Isaac documentation. (https://isaac-sim.github.io/IsaacLab/main/source/how-to/import_new_asset.html).
Once you have the USDA file, you can manually edit the structure so it matches the same format we have for SMPL, or you can edit the robots.py file to access the correct prims (both should work).

I don't think you need to add specific collision filtering, it should automatically filter collisions from parent/child nodes and the rest should be enabled/disabled based on whether you want any collisions at all.

I believe the G1 should also work with collisions. We simply used the settings we found are common in other repos for consistency.