NVlabs/ProtoMotions #117: How to add a custom robot (ROBOTIS-OP3) and customize observations?

TPODAvia, 2025-07-05:
Thanks for releasing this great framework! I want to integrate a custom humanoid robot (ROBOTIS-OP3) into ProtoMotions to train policies on it. I would like to know the recommended steps to start with.

- How should I add the OP3 Robot into the project?
- Are there any specific conventions in defining DOFs, actuators, and sensors in ProtoMotions that I should follow to ensure compatibility?
- How can I configure which observation modalities are available to the policy (e.g., joint positions, velocities, IMU, depth camera)?

It would be great if there was some kind of README instruction that could be easily followed.

tesslerc, 2025-08-07:
Sorry for the delayed response.

1. To add the robot you can follow the format in `protomotions/config/robot/`. Each robot has a config file defining how it should behave. It also specifies the mjcf/urdf/usd files corresponding to that robot.
2. We don't require any specific configurations. We rely on Isaac/Genesis to load the robot models.
3. By default we have the proprioception which is joint positions, rotations, and the character's height. This can be edited to add/remove features. You can then add terrain obs in the config as seen [here](https://github.com/NVlabs/ProtoMotions/blob/9f8caaf114279f15ee4d29eb04fb9f9a754c50ea/protomotions/config/exp/path_follower_amp_mlp.yaml#L7).

eadwu, 2025-10-31:
Along these lines how did you convert the SMPL [-X] rigs to be mjcf/usd physically compatible? Looking to add a few more "bones".

Looking at the XML there seems to be a lot of constant values. Do you have a script to convert from a rig to a base robot before the stiffness/damping tinkering?

closed 2026-01-14 by tesslerc