NVlabs/ProtoMotions #118: How to use text driven control

Greatsjk, 2025-07-08:
First, add the flag "--create-text-embeddings" does not work for python data/scripts/package_motion_lib.py, this python file does not have this function or parameter.
Second, how to use text-driven control? When I run protomotions/eval_agent.py with the follow ars:
                "+robot=smpl",
                "+simulator=isaacgym",
                "+opt=masked_mimic/tasks/user_control",
                "+checkpoint=data/pretrained_models/masked_mimic/smpl/last.ckpt",
                "+terrain=flat",
                "+motion_file=data/motions/humanml3dmotion.pt"
I find this error:
n '_dummy_empty_config_': Could not find 'opt/masked_mimic/tasks/user_control'

Available options in 'opt':
	slurm_autoresume_srun
	wandb
Config search path:
	provider=hydra, path=pkg://hydra.conf
	provider=main, path=file:///home/kale/workspace/ProtoMotions/protomotions/config
	provider=schema, path=structured://
  File "/home/kale/workspace/ProtoMotions/protomotions/eval_agent.py", line 99, in <module>
    main()
hydra.errors.MissingConfigException: In '_dummy_empty_config_': Could not find 'opt/masked_mimic/tasks/user_control'

Available options in 'opt':
	slurm_autoresume_srun
	wandb
Config search path:
	provider=hydra, path=pkg://hydra.conf
	provider=main, path=file:///home/kale/workspace/ProtoMotions/protomotions/config
	provider=schema, path=structured://

Greatsjk, 2025-07-09:
I want to report a logging bug: In tensorboard and wandb, this code only reports the value on the current step and will not plot a curve during the training step.

tesslerc, 2025-07-20:
Change the X axis to epoch instead of steps.
It will show the full curve.

I'll try find a way to set the default x axis from code

tesslerc, 2025-07-20:
Regarding text control, we've removed this in v2 of the codebase.
You can find examples in v1 for how to add it back.
We might bright it back in the future, if we can find a more natural way to integrate it without over complicating the codebase.