NVlabs/ProtoMotions #124: Details of constraints setting for masked mimic

minolta001, 2025-07-30:
Hi. I am evaluating some maskedmimic policies on H1-2. At current stage, I only use the constraint `pelvis_position` and `head_position` as the testing constraints. However, I don't understand how `masked_mimic_time_gap_probability` and `masked_mimic_repeat_mask_probability` impact the evaluation, do you mind provide more details about them?

One point I figured out is, if I set `masked_mimic_time_gap_probability` to 1, then the red color target motion trajectory will become invisible. Does this mean it will be forced to mask all motion trajectory, but only make the goal points (e.g., pelvis) visible?

Thanks in advance!

tesslerc, 2025-08-07:
The masking mechanism has several knobs you can control.
- `masked_mimic_repeat_mask_probability` determines the probability of reusing the previous mask across steps. This is not used when you manually set the `masked_mimic_fixed_conditioning` as the fixed-conditioning overrides the random sampling.
- `masked_mimic_time_gap_probability` this works in cohort with the `time_gap_mask_[max/min]_steps`. Each time a time-gap is sampled, it will introduce a random number of steps during which conditioning is not provided.

The knobs above refer to short-horizon conditioning. Typically poses within the next 10 frames.

- `target_pose_visible_prob` considers a long horizon objective. Up to 1 target pose can be active. When a target pose is sampled it will same the pose itself and the time to reach it. Upon reaching it, we sample a new target pose. The target pose operates independently from the near-term masking mechanism.

Tracking objective is typically near-term. Disable the time-gap and disable the target pose.
For inbetweening you typically don't want any near term conditioning. You want long-term conditioning and to allow the model to generate. In that case you want to introduce maximal time-gap (probability 1 and the min/max values to be high. And to set the target pose prob to 1).

I hope this is helpful!

tesslerc, 2025-12-11:
Closing. Feel free to re-open if still relevant.

closed 2025-12-11 by tesslerc